opgm/openpilot #111: GM cars without ACC using CC spamming have OP engaged tied to button presses and not PCM status

garrettpall, 2024-04-23:
Panda controls allowed is set to PCM status as it probably should.

Not have OP tied to PCM status causes:
- LKAS errors because of panda rejecting messages
- OP not disengaging when OP sends a cancel button press

Having OP tied to PCM status causes:
- initialize_v_cruise to need modification to record last valid button press and use that for initialization (when setting off of PCM status, the button press already ended before initialize_v_cruise runs)
- update_v_cruise to need `self.CP.pcmCruise and not "Redneck CC"`

nworb-cire, 2024-08-10:
Let's tie OP engagement status to the PCM. It's a little bit messy for now, but we may be able to clean it up a little bit as time goes on to make the diff minimal.

Can you re-open a PR with the required code modifications?

garrettpall, 2024-08-22:
https://github.com/opgm/openpilot/pull/120

closed 2024-08-22 by garrettpall